playfultechnology/audio-guestbook #127: Read dial numbers

pmiv, 2023-11-10:
Is it possible to detect the numbers like if the user makes a call and do something then?
Thank you, great projet.

zaff0, 2024-03-01:
Possible, with a lot of coding. The rotary dial, in my phone at least, has two contacts, one that goes closed when the dial starts it opens again when the rotation finished, and the second that pulses for every digit in the rotation, 10 times for the number 0.

Handerson91, 2024-03-06:
Alister has another tutorial on his youtube for using a rotary phone as an escape room prop.  He talks through the process but the code isn't publicly available like this one is.  I'd love to try and combine the two to get this feature working but I don't know when I'll have time to try as my wedding is in a month so getting the basic functions working is the priority for now!

zaff0, 2024-03-06:
> Alister has another tutorial on his youtube for using a rotary phone as an escape room prop. He talks through the process but the code isn't publicly available like this one is. I'd love to try and [...]

Best of luck with your wedding 🙏🏽.
I still have a couple of months.
yeah best to get the basics working correctly. you can always perfect it after the honeymoon 😁

alastaira, 2024-03-25:
See https://youtu.be/cZ2rHqBXO1s?si=kABzR09d1tE-B2Ip

closed 2024-03-25 by alastaira