playfultechnology/audio-guestbook #129: Greeting will not play

brynlcdjs, 2023-11-20:
Hello I have tried so many different things at this point to get my guestbooks (I have 3, 2 of them have this issue 1 somehow works fine) to work such as replacing the greeting and guest folders, making sure the greeting file is .wav, and even formatting my micro SD cards with zero luck. No matter what I do, when I pick up the handset I get a beep then it starts recording. I need to get these back to the point of playing the pre recorded greeting and am really at the end of my rope trying to figure anything out. Any advice at all would be massively appreciated.

Teflon843, 2024-01-11:
Hello, did you ever resolve your issue?

brynlcdjs, 2024-01-17:
Yes I finally heard back from the company who we purchased from, they sent the original file which is working. I guess it needed to be specifically labeled GREETING.mp3 even though I tried changing another one to that and it didn't work. So I am not 100% sure what the exact issue was but it is working now.