playfultechnology/audio-guestbook #132: MTP_Teensy.h: No such file or directory

Patrick838e, 2023-11-23:
Arduino: 1.8.9 (Windows 10), TD: 1.58, Board: "Teensy 4.0, Serial, 150 MHz, Faster, German, Normal"

Several libraries were found for "SD.h".
C:\Users\rosen\OneDrive\Documents\Arduino\audio-guestbook\audio-guestbook.ino:30:10: fatal error: MTP_Teensy.h: No such file or directory
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
   30 | #include <MTP_Teensy.h>
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
      | ^~~~~~~~~~~~~~
compilation terminated.

Error compiling for board Teensy 4.0.

This report would be more detailed if the option
"Verbatim output during compilation"
in File -> Preferences would be activated.

PulsarDevLabs, 2024-01-19:
I have this same error. Were you able to figure it out?

jaredml, 2024-02-03:
you will have to download the repository from the mtp-teensy github page. you will then have to copy the whole extracted folder to your teensy libraries folder. The folder path should look something like this C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries.

MTP Teensy page: https://github.com/KurtE/MTP_Teensy

Imutchy, 2024-08-02:
I too have this error. Not IT savvy at all 46yo newby. What repository am I downloading from your link? There's loads of folders and code???? Thanks.

Imutchy, 2024-08-02:
Arduino: 1.8.19 (Windows 7), TD: 1.59, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

Multiple libraries were found for "SD.h"

In file included from C:\Users\Sarah\Documents\Arduino\hardware\teensy\avr\libraries\Audio/Audio.h:129,

 Used: C:\Users\Sarah\Documents\Arduino\hardware\teensy\avr\libraries\SD

 Not used: C:\Users\Sarah\Documents\Arduino\libraries\SD

                 from C:\Users\Sarah\Documents\Arduino\sketch_jul28a\sketch_jul28a.ino:25:

C:\Users\Sarah\Documents\Arduino\hardware\teensy\avr\libraries\Audio/play_serialflash_raw.h:33:10: fatal error: SerialFlash.h: No such file or directory

   33 | #include <SerialFlash.h> // github.com/PaulStoffregen/SerialFlash/blob/master/SerialFlash.h

      |          ^~~~~~~~~~~~~~~

compilation terminated.

Error compiling for board Teensy 4.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jaredml, 2024-08-04:
> I too have this error. Not IT savvy at all 46yo newby. What repository am I downloading from your link? There's loads of folders and code???? Thanks.

You are downloading the whole repository. You will need to unzip the GitHub project and then put it into the folder I specified.

Imutchy, 2024-08-05:
Sorry Jared, coding skills are nil. Please be patient. I go to the project tab but it is empty? There are folders in the code tab? Where and what exactly do I do? Thank you

jwhitt1990, 2024-09-17:
hi jared i too am working on this project and have hit this same snag

w1802464, 2024-10-03:
I found a fix.

Prequisite:
In Arduino IDE:
Go to File > Preferences
Make sure Sketchbook location is something like this:
"C:\Users\Sarah\Documents\Arduino"

Next:
Download the github zip library mentioned by jaredml
*You do not need to move or unzip files

Then:
In Arduino IDE:
Go to Sketch > Include Library > Add .ZIP Library 
Then add the library that was downloaded. 

That's it. I hope that helps.