playfultechnology/audio-guestbook #133: Not recording to the SD Card

Teflon843, 2024-01-11:
Hey guys, I ran into several issues with this code but was able to figure it out over time. I have been stuck on this last issue for several weeks now. Maybe someone here can shed a little light on the issue.

When I leave a message, nothing is recorded. The file is written to the SD Card but there is no sound when you play it. and when you click on the .wav file it plays but no sound.

Everything appears to be working properly through the serial monitor. Just nothing when it comes to sound on the SD Card. Again it writes the file name, date and time, but doesn't play any sound from the mic recording.

Ive disconnected the mic from the audio shield to see what would happen and it does the same thing. Ive used 3 other old phone mics and still the same thing. It couldn't be all the mics are completely dead, or could it be. Even if damaged I would expect to get some type of noise from the mic.

Any suggestions?

Casaluifred, 2024-01-12:
So You use the Original Microphone?
In my case it works, but very silent.

I use thus one, it works perfekt.
The plastic part around ist printed with my 3D-printer, it works also without.

[image]

Casaluifred, 2024-01-12:
[image]

Teflon843, 2024-01-12:
Yes used the original mics and another mic. I don’t believe the mic is the issue. I believe it is the code. 

I’m not sure what the problem is. Files are been written to the sdcard but there is no recording.

Casaluifred, 2024-01-12:
Have you also tried swapping the two cables on the microphone?
With a small microphone, it does matter how it is connected.
I can provide you with my code. It's a mix of playfultechnology's and DD4WH's with a few changes. 
But I didn't change the audio part.
I think it's a hardware problem.

ninja-warrior, 2024-02-02:
Sorry for the Offtopic:

Could you please provide a link or file to print the plastic housing for the microphone? This would be awesome!

Brent3186, 2024-02-07:
> Have you also tried swapping the two cables on the microphone? With a small microphone, it does matter how it is connected. I can provide you with my code. It's a mix of playfultechnology's and DD4WH's with a few changes. But I didn't change the audio part. I think it's a hardware problem.

Just starting this project, would you mind sending a link  to your code

Teflon843, 2024-02-20:
Hey Casaluifred could you please provide the code for me to try? Ive tried using the mic on the phone itself. Ive tried electrek mics and no sound at all.

Teflon843, 2024-02-20:
> Have you also tried swapping the two cables on the microphone? With a small microphone, it does matter how it is connected. I can provide you with my code. It's a mix of playfultechnology's and DD4WH's with a few changes. But I didn't change the audio part. I think it's a hardware problem.

Please any help you can provide would be great.

Kenitilas, 2024-05-19:
I am also not really happy with my mic quality. I am using the original mic and it is working but it is a bit crisp...
@Casaluifred did you just used the mic DD4WH recommended? Here in Germany these models are quite hard to find...