playfultechnology/audio-guestbook #137: Power issue?? Works fine only when plugged into computer!

Handerson91, 2024-03-06:
Hi all, I'm really hoping someone can help me! I'm so close to getting this thing working!

I've worked through a bunch of errors and issues thanks to all the other comments and now when I have the phone pulled apart and connected by USB to the computer, it works great! I managed to edit the code to not only play back the most recent recording but to be able to skip back through the 5 most recent recordings (see my comment on #62 for details). 

However, as soon as I unplug from the CPU and try to use any other power source, it doesn't work.  The red LED flashes on and off slowly and nothing works.  At first I thought it was the battery pack - I've tried to attach a 3xAA battery pack to the VIN and G ports  and upon unplugging and testing with batteries, it wouldn't work (the LED flashing is dimmer than when it's plugged into the CPU).  Next I tried to plug it into a USB battery pack and while the LED flashing was the same as when plugged in, there was still no function!  

If anyone has any suggestions or can tell me where I'm going wrong, I'd be SO appreciative! I'm just under a month out of the wedding and I really want to be able to have it working as a self-contained unit and not plugged into a laptop! I don't trust drunken guests that much!

Thanks in advance

MonkeyHanger95, 2024-09-22:
Hi Handerson91, where you able figure this one out? I'm in the same situation as you were, works fine however I want contain the board and power supply in an external box. 

What did you go with in the end for the batter supply? I was planning on using 3xAA batteries, I have a 4 slot aa supply but think the voltage is too high for what the VIn can receive.

Any help would be appreciated. Thanks.

Handerson91, 2024-09-23:
It's been a few months but someone replied to one of my comments and helped me fix the issue.  I can't find the thread now so maybe it's been deleted.  Not sure if you're doing what I was doing but I thought the button on the Teensy was a restart button but it's a full reset.  From memory, you have to configure the code onto the teensy plugged into the computer and once it's done, immediately unplug. Don't test it by picking up the receiver or it holds it on a loop waiting for input from the computer even after you disconnect it.  If you press the button on the teensy, it wipes the code stored on it and so there's nothing for it to run.  Mine worked fine after that. 

Pretty sure I went with a 3xAA battery pack.  I can't access it to check at the moment though.  I was able to get it all contained within the body of the phone so it was completely portable. Good luck!

MonkeyHanger95, 2024-09-26:
That's exactly what I was doing... I didn't realise it was being wiped and thought that was just a reset button as you said.

I've gone for 3xAA batteries and it works fine (the connection to the 5v header is currently loose & doing my head in but i'll fix that). 

Thanks mate, appreciate the help