playfultechnology/audio-guestbook #139: Greeting message too quiet

Sopzeh, 2024-03-28:
Hi,

We struggled through many errors and now we have a fully functioning system.
However, our greeting message is too quiet, and we can't figure out which parameter to change to adjust this.
The beep volume is fine.

Thanks!

milesnp, 2024-04-05:
Hi @Sopzeh, I hope I'm not too late to help. You can adjust the volume of the greeting by increasing the `1.0f` value on line 117:
```
mixer.gain(1, 1.0f);
```

Sopzeh, 2024-04-05:
Hi, not too late quite! We tried that before posting, it didn't achieve anything. I'm wondering if we are limited by the physical capability of the speaker.
I posted on the Teensy forum and received an updated version of the programme with improvements including a more specific greeting message audio control. I've sent it to the couple but not sure if they implemented it or not yet.
Appreciate your help!

TateWalker, 2024-06-14:
@Sopzeh do you have the link to that new program? I'm running into the same issue. I'll post my findings if you send it over

Sopzeh, 2024-06-15:
Hi,

Do you still need the link? I can't see your comment on Github

TateWalker, 2024-06-20:
It was a wiring issue. Connected the amplifier cable to the line out slot of the audio shield and it fixed the issue. Wired microphone straight to the shield instead of going through the terminal block and that amplified that and removed noise on recordings. Thanks anyway