playfultechnology/audio-guestbook #146: Greeting and beep unclear/muffled

bullfrogdog, 2024-07-05:
My phone is recording messages perfectly, but when the handset is picked up, the greeting and beep are extremely muffled. You can tell that the greeting and beep are playing whenever the phone is picked up because the voice message doesn’t start recording until the greeting and beep are played and you can hear mumbles in the speaker that is supposed to be the greeting and beep, but it is unclear. Everything else is working perfectly.

coldkevin, 2024-07-28:
i had to switch the audio tweaks back to normal from 'bigger blocks'

everything was working fine, tried the bigger blocks tweak and everything sounded like garbage. moved it back to normal and the sound was back to normal

JordyForNow, 2024-08-27:
I had the same problem, I think the tweaks are not necesary anymore since the new arduino IDE update