playfultechnology/audio-guestbook #147: MTP_Teensy error need to select USB Type.

Adrian-Haber, 2024-07-12:
When compiling the latest version of the code I get the following error. Not sure how to fix it.

In file included from C:\Users\haber\Downloads\audio-guestbook-main\audio-guestbook\audio-guestbook.ino:30:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MTP_Teensy\src/MTP_Teensy.h:32:2: error: #error "You need to select USB Type: 'MTP Disk (Experimental)' or 'Serial + MTP Disk (Experimental)'"
   32 | #error "You need to select USB Type: 'MTP Disk (Experimental)' or 'Serial + MTP Disk (Experimental)'"
      |  ^~~~~

newsman1, 2024-07-14:
I'm having the same issue

fritzimollo, 2024-07-16:
I'm having the same issue. Where can you adjust this?